adafruit/Adafruit_CircuitPython_PN532 #3: MemoryError: memory allocation failed

andreykats, 2018-08-27:
main.py output:
Traceback (most recent call last):
  File "main.py", line 15, in <module>
  File "adafruit_pn532.py", line 53, in <module>
MemoryError: memory allocation failed, allocating 80 bytes

Is this library very memory heavy? I can't seem to get it running on my Trinket M0 along with the Adafruit.hid Keyboard library. The simple sample code does seem to work just fine.

ladyada, 2018-08-27:
yeah this is a very complex chip. we only tested it on an M4 and raspi - maybe try without HID library and see if towkrs then

ladyada, 2018-08-27:
(you're using the .mpy right?)

andreykats, 2018-08-27:
Thanks for creating this library by the way. Yea, I'm using the mpy. Its working just fine without the HID library which is good but what I was really hoping to make is a NFC to USB device using your little Trinket M0.

ladyada, 2018-08-27:
if you're feeling up for it you might be able to split the pn532.py into 4 pices - the main pn532, one for i2c, one for spi and one for uart...then you would only import one!

caternuson, 2018-08-27:
just to verify, *.mpy version of all:
```python
Adafruit CircuitPython 3.0.1 on 2018-08-21; Adafruit Trinket M0 with samd21e18
>>> from adafruit_hid.keyboard import Keyboard
>>> import adafruit_pn532
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "adafruit_pn532.py", line 468, in <module>
MemoryError: memory allocation failed, allocating 584 bytes
>>> 
```

caternuson, 2018-08-28:
just documenting that this also doesn't work:
```python
Adafruit CircuitPython 3.0.1 on 2018-08-21; Adafruit Trinket M0 with samd21e18
>>> from adafruit_hid.keyboard import Keyboard
>>> from adafruit_pn532 import PN532_I2C
>>> import board, busio
>>> i2c = busio.I2C(board.SCL, board.SDA)
MemoryError: memory allocation failed, allocating 384 bytes
>>> 
```
was worth a try though.

ladyada, 2018-08-28:
chekcin, you split up the files?

caternuson, 2018-08-28:
yep. here you go - #6 

here's a quick I2C test which includes the HID import to verify it now works:
```python
import board, busio
from adafruit_hid.keyboard import Keyboard
from adafruit_pn532.i2c import PN532_I2C

i2c = busio.I2C(board.SCL, board.SDA)

pn532 = PN532_I2C(i2c)

pn532.SAM_configuration()

print('Waiting for RFID/NFC card...')
while True:
    # Check if a card is available to read
    uid = pn532.read_passive_target(timeout=0.5)
    print('.', end="", flush=True)
    # Try again if no card is available.
    if uid is None:
        continue
    print('Found card with UID:', [hex(i) for i in uid])
```

I mpy-cross'd everything before copying over to CP board. Tested on Trinket M0:
```python
Adafruit CircuitPython 3.0.1 on 2018-08-21; Adafruit Trinket M0 with samd21e18
>>> import pn532_i2c_test
Waiting for RFID/NFC card...
....Found card with UID: ['0xf1', '0x4a', '0xb5', '0xfb']
.Found card with UID: ['0xf1', '0x4a', '0xb5', '0xfb']
.Found card with UID: ['0xf1', '0x4a', '0xb5', '0xfb']
.Found card with UID: ['0xf1', '0x4a', '0xb5', '0xfb']
```

ladyada, 2018-08-28:
@andreykats wanna try it?

caternuson, 2018-08-28:
@andreykats Yes! Please do. If you need help figuring out how to mpy-cross the *.py files, let us know.

caternuson, 2018-08-28:
To get this files, clone the repo from here:
https://github.com/caternuson/Adafruit_CircuitPython_PN532
and then switch to the branch with the changes:
`git checkout iss3_refactor`

Then, get a precompiled exectuable of mpy-cross from here:
https://github.com/adafruit/circuitpython/releases/
Get the one for your operating system.

Generate a .mpy for each of the .py files by running mpy-cross. For example:
`mpy-cross adafruit_pn532.py`
Do the same thing for the other .py files.

Delete the existing library file from your CP board - CIRCUITPYTHON/lib/adafruit_pn532.mpy

Create a new folder on your CP board - CIRCUITPYTHON/lib/adafruit_pn532

Put all of the .mpy files in there.

Then you should be ready to test!

andreykats, 2018-08-29:
@caternuson Sorry, this is a little unfamiliar to me. I download the mpy-cross for my os and tried run `./mpy-cross adafruit_pn532.py` and `mpy-cross adafruit_pn532.py` but getting "command not found". Am I missing a step?

caternuson, 2018-08-29:
It sounds like it's not finding the mpy-cross program. What OS are you on?

caternuson, 2018-08-29:
Did you rename the file? It looks like it downloads as `mpy-cross-3.x-macos-high-sierra`. You may also need to change it to be executable:
```
chmod +x mpy-cross-3.x-macos-high-sierra
```

caternuson, 2018-09-12:
@ladyada Finally getting back to this. Do you still want to go this refactor route? If so I'll work on changing the examples to match and make Travis happy so #6 can be merged.

ladyada, 2018-09-12:
yep im into it! want me to test?

caternuson, 2018-09-12:
that'd be super!

ladyada, 2018-09-16:
should be resolved in latest commit, zoom!

ladyada, 2018-09-16:
& released! check v2.0.0 and reopen if still an issue

closed 2018-09-16 by ladyada