adafruit/Adafruit_CircuitPython_PN532 #4: incorrect import path for examples

caternuson, 2018-08-27:
Looks like examples are still using development environment setup?

https://github.com/adafruit/Adafruit_CircuitPython_PN532/blob/master/examples/pn532_simpletest.py#L11
https://github.com/adafruit/Adafruit_CircuitPython_PN532/blob/master/examples/readwrite_ntag2xx.py#L31

```python
Adafruit CircuitPython 3.0.1 on 2018-08-21; Adafruit Metro M4 Express with samd51j19
>>> from Adafruit_Circuitpython_PN532 import adafruit_pn532
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: no module named 'Adafruit_Circuitpython_PN532'
>>> 
```

ladyada, 2018-08-27:
woops shoot. wanna send in a PR?

caternuson, 2018-08-28:
easy mcpeasy. there yuh go.

ladyada, 2018-08-28:
huzzah!

closed 2018-08-28 by ladyada