adafruit/Adafruit_CircuitPython_PN532 #7: Add @property style access

caternuson, 2018-09-19:
Provide easier access to functionality via `@property` style access.

kattni, 2018-12-03:
@caternuson Is this something you're still interested in doing? If not, please consider providing a more detailed description and unassign yourself so we can open up the issue to others. Thanks!

caternuson, 2018-12-03:
Still generally interested. There's nothing currently broken, so this is lower priority. If anyone else wants to take this on, please do. Some of these are obvious, like change:
```python
get_firmware_version()
```
to an `@property` called something like `firmware_version`. With others, not so obvious, since there are some values that are being passed in as parameters.

FoamyGuy, 2020-02-29:
I'm going to work on this over the weekend.

FoamyGuy, 2020-02-29:
For reference later there are some code snippets in this [learn guide section](https://learn.adafruit.com/adafruit-pn532-rfid-nfc/python-circuitpython#circuitpython-and-python-usage-5-20) that would need to get updated to use the new property instead of `get_firmware_version()`

FoamyGuy, 2020-03-18:
resolved with #30

closed 2020-03-18 by FoamyGuy