adafruit/Adafruit_CircuitPython_PN532 #12: pn532_simpletest.py - UART - busio.UART not supported on this platform. Please use pyserial instead.

chuenniger, 2019-01-18:
Hi

I´ve intalled the libs and setup all devices but when I debug the code I get an error

```python
ssh://pi@sitelog-raspberrypi-1:22/usr/bin/python3 -u /home/pi/.pycharm_helpers/pydev/pydevd.py --multiproc --qt-support=auto --client 0.0.0.0 --port 42697 --file /home/pi/python/slac-client/tests_sitelog/rfidreader_new/rfid_uart.py
pydev debugger: process 1226 is connecting

Connected to pydev debugger (build 183.5153.39)
Traceback (most recent call last):
  File "/home/pi/.pycharm_helpers/pydev/pydevd.py", line 1741, in <module>
    main()
  File "/home/pi/.pycharm_helpers/pydev/pydevd.py", line 1735, in main
    globals = debugger.run(setup['file'], None, None, is_module)
  File "/home/pi/.pycharm_helpers/pydev/pydevd.py", line 1135, in run
    pydev_imports.execfile(file, globals, locals)  # execute the script
  File "/home/pi/.pycharm_helpers/pydev/_pydev_imps/_pydev_execfile.py", line 18, in execfile
    exec(compile(contents+"\n", file, 'exec'), glob, loc)
  File "/home/pi/python/slac-client/tests_sitelog/rfidreader_new/rfid_uart.py", line 38, in <module>
    uart = busio.UART(board.TXD, board.RXD, baudrate=115200, timeout=100)
  File "/home/pi/.local/lib/python3.5/site-packages/busio.py", line 158, in __init__
    from machine import UART as _UART
ImportError: No module named 'machine'

Process finished with exit code 1
```

The code is from the file `examples/pn532_simpletest.py`

```python
"""
This example shows connecting to the PN532 with I2C (requires clock
stretching support), SPI, or UART. SPI is best, it uses the most pins but
is the most reliable and universally supported.
After initialization, try waving various 13.56MHz RFID cards over it!
"""

import board
import busio
from digitalio import DigitalInOut
#
# NOTE: pick the import that matches the interface being used
#
# from adafruit_pn532.i2c import PN532_I2C
# from adafruit_pn532.spi import PN532_SPI
from adafruit_pn532.uart import PN532_UART

# I2C connection:
# i2c = busio.I2C(board.SCL, board.SDA)

# Non-hardware
# pn532 = PN532_I2C(i2c, debug=False)

# With I2C, we recommend connecting RSTPD_N (reset) to a digital pin for manual
# harware reset
# reset_pin = DigitalInOut(board.D6)
# On Raspberry Pi, you must also connect a pin to P32 "H_Request" for hardware
# wakeup! this means we don't need to do the I2C clock-stretch thing
# req_pin = DigitalInOut(board.D12)
# pn532 = PN532_I2C(i2c, debug=False, reset=reset_pin, req=req_pin)

# SPI connection:
# spi = busio.SPI(board.SCK, board.MOSI, board.MISO)
# cs_pin = DigitalInOut(board.D5)
# pn532 = PN532_SPI(spi, cs_pin, debug=False)

# UART connection
uart = busio.UART(board.TXD, board.RXD, baudrate=115200, timeout=100)
pn532 = PN532_UART(uart, debug=False)

ic, ver, rev, support = pn532.get_firmware_version()
print('Found PN532 with firmware version: {0}.{1}'.format(ver, rev))

# Configure PN532 to communicate with MiFare cards
pn532.SAM_configuration()

print('Waiting for RFID/NFC card...')
while True:
    # Check if a card is available to read
    uid = pn532.read_passive_target(timeout=0.5)
    print('.', end="", flush=True)
    # Try again if no card is available.
    if uid is None:
        continue
    print('Found card with UID:', [hex(i) for i in uid])
```

The jumpers were set correctly. (No jumpers for UART)

My setup:
 - Raspberry PI 3 B
 - PN532 Breakout Board: https://www.adafruit.com/product/364

caternuson, 2019-01-18:
On the Pi, you will need to use pyserial. See here for an example that discusses `busio.UART` vs. `serial.Serial`:
https://learn.adafruit.com/circuitpython-on-raspberrypi-linux/uart-serial#run-that-code-8-28

chuenniger, 2019-01-21:
thnq :)

closed 2019-01-21 by chuenniger