adafruit/Adafruit_CircuitPython_PN532 #13: Response checksum did not match expected value

Taewan-P, 2019-01-19:
Hello!
I am trying to use pn532_simpletest.py for the pn532 module with raspberry pi 3 B+. I am using python 3.5.3 and when you run the sample code and reads some cards, it would raise an RuntimeError like this 

```
File "pn532_simpletest.py", line 50, in <module>
uid = pn532.read_passive_target(timeout=0.5)
File "/usr/local/lib/python3.5/dist-packages/adafruit_circuitpython_pn532-2.0.4.dev0+g915b2e2.d20190119-py3.5.egg/adafruit_pn532/adafruit_pn532.py", line 349, in read_passive_target
File "/usr/local/lib/python3.5/dist-packages/adafruit_circuitpython_pn532-2.0.4.dev0+g915b2e2.d20190119-py3.5.egg/adafruit_pn532/adafruit_pn532.py", line 313, in call_function
File "/usr/local/lib/python3.5/dist-packages/adafruit_circuitpython_pn532-2.0.4.dev0+g915b2e2.d20190119-py3.5.egg/adafruit_pn532/adafruit_pn532.py", line 281, in _read_frame
RuntimeError: ('Response checksum did not match expected value: ', 219)
```

I read the issues from the Adafruid_Python_PN532(The one that is archived now) and found out that another guy was having the same issue. But the comment saids that the issue is solved in this new repo Adafruit_CircuitPython_PN532.

Right now I am working on MiFare Plus cards with the PN532, and some cards do work but some don't and will raise the error above. I am not sure why... Can anyone tell me what's wrong?

ladyada, 2019-01-19:
you can always put it in a try: except block, the PN532 is a little flakey - what interface are you running?

Taewan-P, 2019-01-19:
@ladyada I was just running pn532_simpletest.py from the raspberry pi terminal with the module plugged.

ladyada, 2019-01-19:
what interface

Taewan-P, 2019-01-19:
@ladyada what do you mean what interface? Can you give me an example?

ladyada, 2019-01-19:
wiring, i2c, spi

ladyada, 2019-01-19:
please try SPI - we don't have a guide on using this new library at this time with raspberry pi!

closed 2019-01-19 by ladyada

Taewan-P, 2019-02-15:
@ladyada I just tried spi with sample codes. Remember that I am working on MiFare cards to be recognized? While running the code, spi mode won't even recognize MiFare cards. They don't show a thing... It just keeps on waiting for the card to be tagged.

Taewan-P, 2019-02-15:
Is it normal for the pn532 to not recognize the MiFare cards in spi mode??

ladyada, 2019-02-15:
hiya it should work, we tried it a while ago. check your wiring and your cards

Taewan-P, 2019-02-16:
I tried all the steps again and it worked with spi. When I do nfc-poll and tag a card, it would recognize. **BUT** when I run the example code pn532_simpletest.py, it would throw the same error as i2c.

First, I tried two MiFare cards and they showed up like this:

1.
```Found PN532 with firmware version: 1.6
Waiting for RFID/NFC card...
..........Traceback (most recent call last):
  File "pn532_simpletest.py", line 50, in <module>
    uid = pn532.read_passive_target(timeout=0.5)
  File "/home/pi/.local/lib/python3.5/site-packages/adafruit_pn532/adafruit_pn532.py", line 349, in read_passive_target
    timeout=timeout)
  File "/home/pi/.local/lib/python3.5/site-packages/adafruit_pn532/adafruit_pn532.py", line 313, in call_function
    response = self._read_frame(response_length+2)
  File "/home/pi/.local/lib/python3.5/site-packages/adafruit_pn532/adafruit_pn532.py", line 281, in _read_frame
    raise RuntimeError('Response checksum did not match expected value: ', checksum)
RuntimeError: ('Response checksum did not match expected value: ', 147)
```

And

2. 
```Found PN532 with firmware version: 1.6
Waiting for RFID/NFC card...
.....Traceback (most recent call last):
  File "pn532_simpletest.py", line 50, in <module>
    uid = pn532.read_passive_target(timeout=0.5)
  File "/home/pi/.local/lib/python3.5/site-packages/adafruit_pn532/adafruit_pn532.py", line 349, in read_passive_target
    timeout=timeout)
  File "/home/pi/.local/lib/python3.5/site-packages/adafruit_pn532/adafruit_pn532.py", line 313, in call_function
    response = self._read_frame(response_length+2)
  File "/home/pi/.local/lib/python3.5/site-packages/adafruit_pn532/adafruit_pn532.py", line 281, in _read_frame
    raise RuntimeError('Response checksum did not match expected value: ', checksum)
RuntimeError: ('Response checksum did not match expected value: ', 108)
```

What seems to be the problem??? :( 
(You can see that the error is both same but the checksum value is different.)

Taewan-P, 2019-02-16:
I just deleted the /adafruit_pn532/adafruit_pn532.py code's line 280 and line 281 and the cards would work. What do you have to do with the checksums anyway? I really got curious about the checksum formula in line 279. It seems like the code is comparing the actual checksum and the expected checksum and the cards that I have right now wouldn't work on the expected checksum formula.

Informaticore, 2019-10-01:
@Taewan-P I have the same issue. Yesterday everything was working just fine - now I can not read Mifare cards anymore with that error. I removed the checksum check but I still can not read the UID from the cards. What I experienced as well: whenever I restart the Raspberry Pi (clean boot) and I am running the libnfc `nfc-poll` command, everything works. After I am starting my python script which sends the command for `_COMMAND_INLISTPASSIVETARGET` and try to read a Mifare card, it receives with this checksum error. But after that the whole SPI is broken! Even if I kill the script and try to run `nfc-poll` it returns that RX can not be received. I haven't changed anything on the Device or cabeling. 

Somehow it f**cksup the SPI completely, I can only fix it by rebooting the device again. Normal NFC Cards which come with the modul are readable still. I have no clue what caused this issue as I haven't changed anything.

Taewan-P, 2019-10-01:
@Informaticore Did you try to trace up where the checksum error is raised or whichever error is raised?
That's a bit different compared to my issue. Mine worked perfectly even though I killed my script and run ```nfc-poll```.

Informaticore, 2019-10-01:
@Taewan-P it is fireing the checksum error but with the exact same value for the card. It seems some how, that the SPI is confused (if that is even a thing). I have no Idea what happened - I try to implement the interrupt function so I don't have to poll for the card reading.
The interrupt itself is a strange thing and I got it running some how yesterday - when I tried today, I broke and I can not get it back running again. It just does not read any Mifare cards anymore.

It is like a super weird behavior. I had to copy the Adafruit libs and adjust them to get it running on the Raspberry Pi because the Adafruit only works with the Adafruit Boards (which I really hate btw.). 
But everything worked fine - until it stopped reading Mifare cards and breaks SPI.

It definetly detects the cards but reading the UID fails. I have two different cards and each of them return a different checksum (like 242 and 231) - always! It never changes. 
I am using the NFC Modul V4 (original). I am not able to figure out what the issue really is.

Can it happen, to "confuse" the SPI? Is there a way to "restart" or "flush" SPI?

the-infinity, 2021-03-06:
I just stumpled across this bug using SPI, and I got the source of (my) problems: it's the ATS length. At https://github.com/adafruit/Adafruit_CircuitPython_PN532/blob/master/adafruit_pn532/adafruit_pn532.py#L394 you fix the expected length to 19. This works with Mifare classic, but no ISO14443A-4 card can be read.

The ATS has a max length of 22 bytes. With all the other stuff, this would result in a expected response length of 30. I just created a pull request for this issue.