adafruit/Adafruit_CircuitPython_PN532 #14: Read/write .vcf into mifare cards using pn532 RFID/NFC shield

Nelle96, 2019-02-01:
I was wondering if we can write .vcf files into the cards using PN532 and this library? I'm hoping someone might lend a hand. Really bugged out searching for solutions to this. Thanks!

tgikal, 2019-06-28:
It's all about getting the formatting correct, and having enough room for the data.

Just playing around I have this script working on a RaspberryPi, with a NTAG215 and a NFC vCard app on my phone:

```
import board
import busio
from digitalio import DigitalInOut
from adafruit_pn532.spi import PN532_SPI
# SPI connection:
spi = busio.SPI(board.SCK, board.MOSI, board.MISO)
cs_pin = DigitalInOut(board.D22) # CHANGED TO D22 from D5
pn532 = PN532_SPI(spi, cs_pin, debug=False)
 
# Configure PN532 to communicate with MiFare cards
pn532.SAM_configuration()

print('Waiting for RFID/NFC card!')
while True:
    # Check if a card is available to read
    uid = pn532.read_passive_target(timeout=0.5)
    print('.', end="")

s =  '''text/x-vcardBEGIN:VCARD
VERSION:3.0
N:Gump;Forrest;;Mr.;
FN:Forrest Gump
ORG:Bubba Gump Shrimp Co.
TITLE:Shrimp Man
TEL;WORK:+11115551212
ADR;WORK:;;100 Waters Edge,Baytown;LA;30314;United States of America
EMAIL:forrestgump@example.com
END:VCARD'''

"""
NTAG215 135 pages total
Pages 0,1,2,3 are tag and setup infomation
Pages 4 through 129 should have 504 user bytes available
Last 5 pages are locking information"""
max_page = 129
max_data = 504
print("writing data:", end = "")

# Header telling the reading this is a vCard (at least this what my phone's writer did)
header = bytearray(b'\x03\xff\x01\xaa\xc2\x0c\x00\x00\x01\x98')

termination = bytearray(b'\xFE') # Standard message termination

data =  header + bytearray(s.encode()) + termination

# Clear out the rest of the card
data = data + bytearray(b'\x00'*(max_data-len(data))) 
print(len(data), "bytes of data")

# Write 4 byte pages.
page = 4
for x in range(0,len(data),4):
    pn532.ntag2xx_write_block(page, data[x:x+4])
    page += 1

print('Wrote data in pages 4 to ' + str(page) + ', now trying to read that data:')

r = bytearray()
for x in range(1,max_page):
    r += pn532.ntag2xx_read_block(x)

print(r)
print(r[r.index(b'\x98')+1:r.index(b'\xFE')].decode())
```

kattni, 2019-09-30:
Thanks @tgikal for the example! Closing this issue.

closed 2019-09-30 by kattni