adafruit/Adafruit_CircuitPython_PN532 #19: SPI read_passive_target timeout not working correctly

BraindeadBZH, 2019-02-18:
Hi,

When in SPI mode calling read_passive_target with a timeout, a NFC tag is only detect at the very beginning of the poll. This is not the case with i2c which detects tags through all the duration of the timeout.

tgikal, 2019-07-01:
The SPI and I2C interfaces use an entirely different backend. The problem looks like it's occurring in `adafruit_pn532/spi.py` -> `_wait_ready`.

caternuson, 2019-07-03:
@BraindeadBZH Please try the 2.0.7 version of the library when it becomes available:
https://github.com/adafruit/Adafruit_CircuitPython_PN532/releases/tag/2.0.7

BraindeadBZH, 2019-07-04:
Thank you for fixing the issue. Unfortunately I don't have access to the hardware to test this at the moment. I will let you know if I have the chance to get back to this.

caternuson, 2019-07-09:
OK. Closing for now. We can reopen if you continue to have the same issue.

closed 2019-07-09 by caternuson