adafruit/Adafruit_CircuitPython_PN532 #20: Authenticate with card key

MikeDabrowski, 2019-03-10:
I want to copy other tag which uses key protection. 
I receive this error while trying to authenticate

```python
Traceback (most recent call last):
  File "/home/pi/nfc/rpi-nfc/write.py", line 67, in <module>
    o = pn532.mifare_classic_authenticate_block(uid, 14, MIFARE_CMD_AUTH_B, key)
  File "/home/pi/.local/lib/python3.5/site-packages/adafruit_pn532/adafruit_pn532.py", line 383, in mifare_classic_authenticate_block
    response_length=1)
  File "/home/pi/.local/lib/python3.5/site-packages/adafruit_pn532/adafruit_pn532.py", line 316, in call_function
    raise RuntimeError('Received unexpected command response!')
RuntimeError: Received unexpected command response!
```

Main goal is to first read whole tag that is encrypted using that key and then copy its data onto another and block it with same key. I did it on the mobile app and now want to recreate it in python script.

fezthedev, 2019-05-07:
Could you post your write.py

dherrada, 2019-05-23:
@MikeDabrowski @fezthedev has been unable to replicate this issue. It'll be really hard for us to solve this issue unless you can post the code you have been using.

MikeDabrowski, 2019-05-23:
@fezthedev @dherrada  I'm sorry guys, I've been very busy doing theoretical part of this subject (deadlines), I will be back into this this weekend.

dherrada, 2019-05-25:
@MikeDabrowski No worries. Thanks for responding,

MikeDabrowski, 2019-05-27:
After I gathered more knowledge about those cards I was able to auth 5th block and read/write data there. I also tried modifying key and ended up with whole sector unreadable. 
I think I was using some weird nfc chip at first and that's why it didn't work. With regular mifare classic card it works. I ordered mifare plus and desfire, but they are more complicated and I don't know if this lib can handle them. 
We can close it now, and if I get the same problem I will reopen with all details, that I should've placed at the beginning.

closed 2019-05-27 by MikeDabrowski

dherrada, 2019-05-28:
Ok. From the learn guide, it looks like it’s only set up to read chips that are using the same protocol as the MiFare Classic.

fcrisafulli-dev, 2021-12-31:
For anyone who runs into this problem in the future:

I unpacked my kit 3 years later to do some more projects with the pn532 and came across this same issue. The problem as evaherrada stated above is the type of chip being used on the NFC card itself. The simplest solution is to purchase a pack of NFC cards labeled as MiFare Classic. Those cards should work with the provided examples.

evaherrada, 2022-01-03:
FWIW, I also found it to work fine with my Dangerous things xSIID, which has a Dsruptive SIID in it (13.56MHz ISO14443A & NFC Type 2 chip, whatever the heck that means)