adafruit/Adafruit_CircuitPython_PN532 #23: 'module' object has no attribute implementation

jamiedavid4, 2019-05-11:
trying to run the pn532-simpletest.py file and getting the error 
```Traceback (most recent call last):
  File "pn532_simpletest.py", line 8, in <module>
    import board
  File "/usr/local/lib/python2.7/dist-packages/Adafruit_Blinka-1.2.8-py2.7.egg/board.py", line 32, in <module>
    from adafruit_blinka.agnostic import board_id, detector
  File "/usr/local/lib/python2.7/dist-packages/Adafruit_Blinka-1.2.8-py2.7.egg/adafruit_blinka/agnostic/__init__.py", line 21, in <module>
    implementation = sys.implementation.name
AttributeError: 'module' object has no attribute 'implementation'```

any ideas?

makermelissa, 2019-05-11:
Hi, looks like the issue is you’re only running Python 2. Take a look at this guide: https://learn.adafruit.com/circuitpython-on-raspberrypi-linux/installing-circuitpython-on-raspberry-pi

makermelissa, 2019-05-11:
If you need further help, try the forums or discord.

closed 2019-05-11 by makermelissa