adafruit/Adafruit_CircuitPython_PN532 #34: Failed to detect the PN532 - Rasp Pi Zero with PN532 Shield v1.3

dangell7, 2020-04-12:
I'm using the following:

Shield PN532 - v1.3
Rasp Pi Zero - v1.3

I want to use SPI.

PI Pin# 4 (5v) to 532 5V
PI Pin# 25 (Gnd) to 532 Gnd (There are 2? Does it matter?)
PI Pin# 19 (MOSI) to 532 MOSI
PI Pin# 21 (MISO) to 532 MISO
PI Pin# 23 (SCLK) to 532 SCK
PI Pin# 24 (CE0) to 532 SS

I also soldered the SEL0 and SEL1 "closed"? I assume that means connect them together? Or does that mean open? I couldn't tell as the nomenclature used doesn't distinguish and I dont have enough knowledge.

These is what I gathered from all the wire diagrams and tutorials, although, its quite confusing with each different type of board and version

The error I get is "Failed to detect the PN532"

ladyada, 2020-04-12:
hi this is not the appropriate place for tech support... you can post to forums for tech support :)
wiring info is here
https://learn.adafruit.com/adafruit-pn532-rfid-nfc/python-circuitpython

closed 2020-04-12 by ladyada

FoamyGuy, 2020-04-12:
@dangell7 Can you take a picture of your wiring including SEL0 and SEL1 and post it on the adafruit forums? 

Based on the information on this page: https://learn.adafruit.com/adafruit-pn532-rfid-nfc/python-circuitpython

For use with SPI you need SEL0 set to OFF (the right and the center pins connected together with the jumper). and SEL1 set to ON (the left pin and the center pins connected together with the jumper).

I do not think SEL1 and SEL0 should be connected together at all so you may need to undo that. 

Lastly it is probably better to make a post on the Adafruit Forums here: https://forums.adafruit.com/ You can get help with these types of things better there than on Github Issue.