adafruit/Adafruit_CircuitPython_PN532 #35: IRQ not used?

dunkmann00, 2020-07-10:
I just started using the PN532 CircuitPython library and think it is great! But I was curious why IRQ isn't used at all and can't even be optionally used? I have used the c++ library in the past and in that, the IRQ line is used (just for i2c).

I think it could be useful if the IRQ line's readiness (meaning if it is pulled low) was exposed as a property. I was experimenting with this and could issue a pull request if it is something that would be of interest to the maintainers?

i.e.:
```python
@property
def irq_ready(self):
    if self._irq is None:
        raise RuntimeError("IRQ pin not defined!")
    return self._irq.value == 0
```

ladyada, 2020-07-10:
hi you'd just connect it up to a gpio and check it as a gpio, so its not built into the library

dunkmann00, 2020-07-10:
Gotcha, I thought it might be nice having it built into the library, but that way is fine too!

ladyada, 2020-07-10:
all good, thanks :)

closed 2020-07-10 by ladyada