adafruit/Adafruit_CircuitPython_PN532 #36: Add support to 'listen' for tags

dunkmann00, 2020-07-27:
I found that when using the library as it is, it is not possible to avoid a slight delay while trying to read tags, due to `read_passive_target` needing to have a timeout. This is problematic if you are also trying to simultaneously perform another task, like updating neopixels. This delay prevents the neopixels from looking as smooth as they otherwise could.

I worked on tweaking the library slightly to allow, if needed, the ability to separate sending the listen command to the PN532, from reading a tag. None of the other functions are affected and the `read_passive_target` function is also still available and would still be the 'normal' way to read a tag. This additional functionality could be an option if you wanted a way to listen for a tag, while also avoiding any delay.

The idea would be:
1. Start listening
1. Check if a tag is present (by checking IRQ for instance)
1. If no tag is found, perform other tasks
1. Repeat 2 & 3 until a tag is found
1. Read the UID of the tag

It would be similar to how it is currently done, but since checking IRQ is quite quick, you avoid any delay. This would allow other tasks to be run, basically uninterrupted while you wait for a tag.

Any questions or suggestions are welcome! Let me know what you think!

dunkmann00, 2020-08-17:
@ladyada Have you had a chance to take a look at this?

ladyada, 2020-08-17:
nope been super super busy!

dunkmann00, 2020-08-17:
Okay, no worries!

closed 2020-08-19 by ladyada