adafruit/Adafruit_CircuitPython_PN532 #42: using uart raspi "unicode strings are not supported"

ethugbaby, 2020-09-16:
i am trying to read nfc using a raspi 4b i have pn532 connected through uart 
```
"""
This example shows connecting to the PN532 with I2C (requires clock
stretching support), SPI, or UART. SPI is best, it uses the most pins but
is the most reliable and universally supported.
After initialization, try waving various 13.56MHz RFID cards over it!
"""

import board
import busio
from digitalio import DigitalInOut
import serial
#
# NOTE: pick the import that matches the interface being used
#
#from adafruit_pn532.i2c import PN532_I2C

# from adafruit_pn532.spi import PN532_SPI
from adafruit_pn532.uart import PN532_UART

# I2C connection:
#i2c = busio.I2C(board.SCL, board.SDA)

# Non-hardware
# pn532 = PN532_I2C(i2c, debug=False)

# With I2C, we recommend connecting RSTPD_N (reset) to a digital pin for manual
# harware reset
#reset_pin = DigitalInOut(board.D6)
# On Raspberry Pi, you must also connect a pin to P32 "H_Request" for hardware
# wakeup! this means we don't need to do the I2C clock-stretch thing
#req_pin = DigitalInOut(board.D12)
#pn532 = PN532_I2C(i2c, debug=False, reset=reset_pin, req=req_pin)

# SPI connection:
# spi = busio.SPI(board.SCK, board.MOSI, board.MISO)
# cs_pin = DigitalInOut(board.D5)
# pn532 = PN532_SPI(spi, cs_pin, debug=False)

# UART connection
# uart = busio.UART(board.TX, board.RX, baudrate=115200, timeout=100)

uart = serial.Serial("/dev/ttyS0", baudrate=115200, timeout=10)
pn532 = PN532_UART(uart, debug=False)

ic, ver, rev, support = pn532.firmware_version
print("Found PN532 with firmware version: {0}.{1}".format(ver, rev))

# Configure PN532 to communicate with MiFare cards
pn532.SAM_configuration()

print("Waiting for RFID/NFC card...")
while True:
    # Check if a card is available to read
    uid = pn532.read_passive_target(timeout=0.5)
    print(".", end="")
    # Try again if no card is available.
    if uid is None:
        continue
    print("Found card with UID:", [hex(i) for i in uid])
```
i keep getting this error
`TypeError: unicode strings are not supported, please encode to bytes: 'UU\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'`

tannewt, 2020-09-16:
Please include the full backtrace. It will tell us what line errors in your example code.

ethugbaby, 2020-09-16:
Im not near my pc I remember the trace back pointing to this line `pn532 = PN532_UART(uart, debug=False)` I’ll post the full thing once I’m home

ethugbaby, 2020-09-16:
```Traceback (most recent call last):
  File "/home/pi/Desktop/uart_rftest.py", line 43, in <module>
    pn532 = PN532_UART(uart, debug=False)
  File "/usr/local/lib/python3.7/dist-packages/adafruit_pn532/uart.py", line 55, in __init__
    super().__init__(debug=debug, reset=reset)
  File "/usr/local/lib/python3.7/dist-packages/adafruit_pn532/adafruit_pn532.py", line 195, in __init__
    self._wakeup()
  File "/usr/local/lib/python3.7/dist-packages/adafruit_pn532/uart.py", line 60, in _wakeup
    self.SAM_configuration()
  File "/usr/local/lib/python3.7/dist-packages/adafruit_pn532/adafruit_pn532.py", line 363, in SAM_configuration
    self.call_function(_COMMAND_SAMCONFIGURATION, params=[0x01, 0x14, 0x01])
  File "/usr/local/lib/python3.7/dist-packages/adafruit_pn532/adafruit_pn532.py", line 296, in call_function
    if not self.send_command(command, params=params, timeout=timeout):
  File "/usr/local/lib/python3.7/dist-packages/adafruit_pn532/adafruit_pn532.py", line 317, in send_command
    self._write_frame(data)
  File "/usr/local/lib/python3.7/dist-packages/adafruit_pn532/adafruit_pn532.py", line 249, in _write_frame
    self._write_data(bytes(frame))
  File "/usr/local/lib/python3.7/dist-packages/adafruit_pn532/uart.py", line 85, in _write_data
    "\x55\x55\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00"
  File "/usr/lib/python3/dist-packages/serial/serialposix.py", line 532, in write
    d = to_bytes(data)
  File "/usr/lib/python3/dist-packages/serial/serialutil.py", line 63, in to_bytes
    raise TypeError('unicode strings are not supported, please encode to bytes: {!r}'.format(seq))
TypeError: unicode strings are not supported, please encode to bytes: 'UU\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'```

dunkmann00, 2020-09-17:
#41 fixes this

ethugbaby, 2020-09-17:
ok im kind of a noob but how does adding low power state fix this and how do i implement it

dunkmann00, 2020-09-18:
@ethugbaby It was something that I fixed while adding the low power code. The two aren't related but it just so happens to be in that pull request. 

You could download the code from my [fork](https://github.com/dunkmann00/Adafruit_CircuitPython_PN532/tree/performance-and-sleep), and use that temporarily, until the pull request is merged in order to get it working. Or for that specific error, if you change the string from `Line 85` in `uart.py` from a regular string to a bytestring that should also fix it.

So instead of this: (`uart.py - Line 85`)
`"\x55\x55\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00"`

It should be this:
`b"\x55\x55\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00"`

There were some other problems with the uart code from the current version. Changing this line will fix the error you're getting but I don't remember if any other change was needed to get the code working. The pull request I made is being tested so hopefully it won't be too long before this is all fixed!

dunkmann00, 2020-09-18:
No worries @ethugbaby, I'm glad I was able to help you! If you run into any other problems feel free to open another issue!