adafruit/Adafruit_CircuitPython_PN532 #43: How can I make read_passive_target wait indefinitely until a tag becomes available?

xetum, 2020-10-04:
Other libraries have this possibility. I don't see how in this one. Thanks

FoamyGuy, 2020-10-04:
You can put it inside of a loop such as in the simpletest example here: https://github.com/adafruit/Adafruit_CircuitPython_PN532/blob/master/examples/pn532_simpletest.py

If if it does not find a card it will just keep retrying until one is in range. 

The relavent code in that example is at the bottom:
```python
print("Waiting for RFID/NFC card...")
while True:
    # Check if a card is available to read
    uid = pn532.read_passive_target(timeout=0.5)
    print(".", end="")
    # Try again if no card is available.
    if uid is None:
        continue
    print("Found card with UID:", [hex(i) for i in uid])
```

xetum, 2020-10-04:
Yes I know, I was trying to do it without the polling loop, as with other
libraries

FoamyGuy, 2020-10-04:
https://github.com/adafruit/Adafruit-PN532/blob/d988fb4a28e54efe4de638883ce00858064ddf5f/Adafruit_PN532.h#L173

In the arduino library there is a comment that mentions using a timeout of 0 will make it wait forever. 

Did you try setting `timeout=0` in the CircuitPython one? Maybe it will work the same.

xetum, 2020-10-04:
Ok, I'll try

xetum, 2020-10-05:
No, timeout=0 does polling. I think it could be added to the library, even
more so the arduino version does blocking. Regards, Francesc